wukongdaily/ImmortalWrt-ImageBuilder #4: 请问，以后如果上游更新了新内核或者新版本，工作流要怎么进行更新

MIT-X, 2024-12-31:
一：比如：24.10分支更新了6.6.68或者更高的内核，工作流要怎么更新呢

再或者上游更新到24.10.1、24.10.2、24.10.3……这样以此类推，工作流要怎么去修改呢

wukongdaily, 2024-12-31:
嗯 根据docker的标签来做修改就行了。yml文件中调用的是其实就是docker镜像。具体的地址如下
`https://hub.docker.com/r/immortalwrt/imagebuilder` 点击tags 后，你搜该平台对应的标签就行了。yml中修改一下对应的标签即可。其他不用动。
顺便说一句，24.10 目前还是测试版本。如果追求稳定，就用25.03.4

MIT-X, 2024-12-31:
手工修改确实可行，工作流建议改成全自动拉取docker上游最新版latest，比如拉取最新23.05.X版latest，拉取最新24.10.X版latest

wukongdaily, 2024-12-31:
哦不是哦 我想你可能理解错了。
首先来说，docker只有标签的概念哦，不存在这个标签里 还套了版本一说哦。
其次这个docker本质是imagebuilder工具，比如23.05.4这个标签就代表了 用于专门编译该luci版本=23.05.4的固件。
可对应网页来看：https://firmware-selector.immortalwrt.org/?version=23.05.4&target=x86%2F64&id=generic
总共也没几个版本了。23.05.4 就到头了。24.10后面还属于测试版。
另外你要仔细看hub docker网站，你要表达的意思应该是最新的代码的意思吧，那应该用 这个标签 [x86-64-master](https://hub.docker.com/layers/immortalwrt/imagebuilder/x86-64-master/images/sha256-380e323b684303e6b513470e426bf99773e52436aac130ed9f8e46fe962a1dfc)
形如：`immortalwrt/imagebuilder:x86-64-master`
这应该就是主分支的标签了。同步最新代码这些逻辑，并非是咱们来控制的。你调用这个标签应该就是最新代码了。
这个适合fork我项目的人自己来修改，因为master分支肯定经常改动。我的主项目想保持跟 https://firmware-selector.immortalwrt.org 网站的luci版本一致。尽可能build稳定一些的版本。创意的地方 大家自己修改就好。

closed 2025-01-01 by MIT-X