wukongdaily/ImmortalWrt-ImageBuilder #6: 自带的 cpu freq 怎么去除？

HUiTHEME, 2024-12-31:
如题

wukongdaily, 2025-01-01:
你好，请问具体你指的是哪款路由器呢？
另外我建议不要在编译的时候去除，因为我们自定义的部分并没有这个组件，说明这是编译工具里自带的，有可能是该target必备的，为了确保万无一失。若想去除，建议在 【系统——软件包】 通过搜索 cpu 等关键字（比如cpufreq） 自行移除 它。 这样做比较保险。

HUiTHEME, 2025-01-01:
哇，悟空大佬回复我了哈，是R66S这个设备，通过 https://firmware-selector.immortalwrt.org/ 搜索设备型号，默认就有这个cpu freq。就是问看看，有无什么办法，能在自动生产配置里，去除一些默认自带的。这个【cpu freq】并非openwrt自带项目，可有可无，本着最纯净化的原则，所以来问问您哈

> 你好，请问具体你指的是哪款路由器呢？ 另外我建议不要在编译的时候去除，因为我们自定义的部分并没有这个组件，说明这是编译工具里自带的，有可能是该target必备的，为了确保万无一失。若想去除，建议在 【系统——软件包】 通过搜索 cpu 等关键字（比如cpufreq） 自行移除 它。 这样做比较保险。

wukongdaily, 2025-01-01:
哦 语法是这样的 `PACKAGES="$PACKAGES -luci-i18n-dockerman-zh-cn"` 
看到那个减号了吗

closed 2025-01-01 by wukongdaily

HUiTHEME, 2025-01-01:
> 哦 语法是这样的 `PACKAGES="$PACKAGES -luci-i18n-dockerman-zh-cn"` 看到那个减号了吗

感谢悟空大佬，新年发大财