wukongdaily/ImmortalWrt-ImageBuilder #10: 我想要一个纯净的拨号上网固件

cjlhll, 2025-01-02:
用来做稳定的主路由系统，是不是在这些不要的插件前面加个-就行了？除了这些还有哪些用不到的插件吗？
# 定义所需安装的包列表
PACKAGES=""
PACKAGES="$PACKAGES curl"
PACKAGES="$PACKAGES luci-i18n-diskman-zh-cn"
PACKAGES="$PACKAGES luci-i18n-firewall-zh-cn"
PACKAGES="$PACKAGES luci-i18n-filebrowser-zh-cn"
PACKAGES="$PACKAGES luci-app-argon-config"
PACKAGES="$PACKAGES luci-i18n-argon-config-zh-cn"
PACKAGES="$PACKAGES luci-i18n-opkg-zh-cn"
PACKAGES="$PACKAGES -luci-i18n-ttyd-zh-cn"
PACKAGES="$PACKAGES -luci-i18n-passwall-zh-cn"
PACKAGES="$PACKAGES -luci-app-openclash"
PACKAGES="$PACKAGES -luci-i18n-homeproxy-zh-cn"
PACKAGES="$PACKAGES openssh-sftp-server"
PACKAGES="$PACKAGES -luci-i18n-dockerman-zh-cn"

wukongdaily, 2025-01-02:
那到不必。一般只有不要某些系统组件才会用到减号的语法。
这本身就是新增的。如果不需要哪个，直接删除那行就可以了。如下所示

```
PACKAGES=""
PACKAGES="$PACKAGES curl"
PACKAGES="$PACKAGES luci-i18n-diskman-zh-cn"
PACKAGES="$PACKAGES luci-i18n-firewall-zh-cn"
PACKAGES="$PACKAGES luci-i18n-filebrowser-zh-cn"
PACKAGES="$PACKAGES luci-app-argon-config"
PACKAGES="$PACKAGES luci-i18n-argon-config-zh-cn"
PACKAGES="$PACKAGES luci-i18n-opkg-zh-cn"
PACKAGES="$PACKAGES openssh-sftp-server"
```

cjlhll, 2025-01-02:
> 那到不必。一般只有不要某些系统组件才会用到减号的语法。 这本身就是新增的。如果不需要哪个，直接删除那行就可以了。如下所示
>
> ```
> PACKAGES=""
> PACKAGES="$PACKAGES curl"
> PACKAGES="$PACKAGES luci-i18n-diskman-zh-cn"
> PACKAGES="$PACKAGES luci-i18n-firewall-zh-cn" [...]

明白了，感谢。

closed 2025-01-02 by wukongdaily