wukongdaily/ImmortalWrt-ImageBuilder #13: OrangePi Zero3的固件刷了无法启动

jackyyow1987, 2025-01-04:
OrangePi Zero3的固件刷了红灯常亮无法启动，试了很多次都如此。刷官网的固件正常。不知道什么原因。

wukongdaily, 2025-01-04:
会不会是卡的问题，我这边是正常的。我记得在tg里 还给出了截图呢 
[image]

wukongdaily, 2025-01-04:
[image]
上面的截图就是 刷了你的release中的固件。你是编译了一个2GB的固件是吧。如图所示 zero3 运行正常的呀。
你编译的固件在我这里没问题。不过我的zero3 是4GB内存的版本。不晓得跟内存是否有关系。

jackyyow1987, 2025-01-04:
谢谢解答。抱歉，我是小白，不知道1024是运行内存的意思。我的是1GB的，我重新编译个试试。

jackyyow1987, 2025-01-04:
我的设备是Zero3，1Gb版本，sd卡是闪迪的。我把后台地址改成了192.168.31.2，刚重新编译了1024内存，带docker，烧录后一直红灯，网线连电脑也无法自动获取IP，无法进入后台，手动设置IP，网关DNS后也无法进入后台。官方的Ubuntu openwrt固件都刷过，正常的。能否把你测试正常的1gb固件发下给我，谢谢。

wukongdaily, 2025-01-04:
1GB的zero3 我也是用的你的release中的固件呀。。。另外红灯亮本来就是正常的。immortalwrt 在zero3 运行的时候，默认就是红色的灯常亮的。LED那边还能自己手动关闭，包括绿灯也能设置，都是可控的。
你尝试把SD卡 先格式化一次。然后在烧录

jackyyow1987, 2025-01-04:
问题已经解决，zero3刷好网线连接电脑，不知道为什么无法获取IP，不知道是不是zero3 之前在路由器后台绑定了静态IP（也已经解绑了）。后来尝试直接连路由器，发现路由器直接分配了原来的IP，用这个IP直接登陆上了后台。

wukongdaily, 2025-01-04:
哈哈 上课不注意听讲。视频里讲过的。
我在重复一下该固件编译之前的设计思路
单网口设备默认DHCP模式。就是自动获取ip模式
多网口设备，lan的ip 为`192.168.100.1`
wan 口是DHCP模式

closed 2025-01-05 by wukongdaily

jackyyow1987, 2025-01-10:
> 哈哈 上课不注意听讲。视频里讲过的。 我在重复一下该固件编译之前的设计思路 单网口设备默认DHCP模式。就是自动获取ip模式 多网口设备，lan的ip 为`192.168.100.1` wan 口是DHCP模式

谢谢。确实没注意到这个部分。瞎折腾了很久。