wukongdaily/ImmortalWrt-ImageBuilder #14: 请问如何默认作为旁路由编译

cjlhll, 2025-01-04:
我改默认的ip地址没生效。刷好系统后，手动改ip和旁路由配置，immortalwrt没网络了。我一直都用旁路由，我认为我设置的没问题。

cjlhll, 2025-01-04:
[image]

wukongdaily, 2025-01-04:
我的设计逻辑是 单网口设备默认dhcp模式进行编译。用的时候就想NAS一样，有上级路由器分配的ip地址，用这个地址访问。设置旁路由的时候 就像你截图那样在改成静态ip，但是记得要放上至少一个dns 比如 `223.5.5.5` 这样才可以确保有网络。

而如果你是多网口设备。访问的时候，用192.168.100.1访问。
lan口接电脑，wan口不接。
设置方法跟上面一样的。

wukongdaily, 2025-01-04:
[image]
[image]
[image]

cjlhll, 2025-01-04:
> 我的设计逻辑是 单网口设备默认dhcp模式进行编译。用的时候就想NAS一样，有上级路由器分配的ip地址，用这个地址访问。设置旁路由的时候 就像你截图那样在改成静态ip，但是记得要放上至少一个dns 比如 `223.5.5.5` 这样才可以确保有网络。
> 
> 而如果你是多网口设备。访问的时候，用192.168.100.1访问。 lan口接电脑，wan口不接。 设置方法跟上面一样的。

可以了，是我无脑修改etc/config/network出问题了，好像immortalwrt不能这么设置是吧。只能通过uci去设置。

closed 2025-01-04 by wukongdaily