wukongdaily/ImmortalWrt-ImageBuilder #16: 请问需要添加一个rootfs的格式输出，如何添加？

wander555, 2025-01-06:
请问需要添加一个rootfs的格式输出，如何添加配置文件？
想用在pve的LXC容器中，占用更低

wukongdaily, 2025-01-06:
这需要改 x86-64/imm.config文件的配置。具体来说要放开这句，你在文件里搜一下。这里改成y
`CONFIG_TARGET_ROOTFS_CPIOGZ=y`

另外yml中，
`build-x86-64-23.05.4.yml `
```
- name: Upload ImmortWrt as release assets
        uses: softprops/action-gh-release@v2.1.0
        with:
          tag_name: Autobuild-x86-64
          files: |
            ${{ github.workspace }}/*.img.gz.sha256
            ${{ github.workspace }}/bin/targets/x86/64/*.img.gz
          token: ${{ secrets.GITHUB_TOKEN }}
        env:
          GITHUB_TOKEN: ${{ secrets.GITHUB_TOKEN }}
```

wander555, 2025-01-06:
> 这需要改 x86-64/imm.config文件的配置。具体来说要放开这句，你在文件里搜一下。这里改成y `CONFIG_TARGET_ROOTFS_CPIOGZ=y`
>
> 另外yml中， `build-x86-64-23.05.4.yml `
>
> ```
> - name: Upload ImmortWrt as release assets
>         uses: [...]

感谢！我去试试

closed 2025-01-06 by wukongdaily