wukongdaily/ImmortalWrt-ImageBuilder #18: 24.10 X86 编译出错

androidcn, 2025-01-06:
` * opkg_install_cmd: Cannot install package ppp-mod-pppoe.
Configuring luci-light.
Configuring luci.
Configuring default-settings.
Configuring luci-i18n-base-zh-cn.
Configuring default-settings-chn.
Configuring luci-i18n-ttyd-zh-cn.
Configuring fdisk.
Configuring urngd.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:161: _call_image] Error 2
make: *** [Makefile:337: image] Error 2
2025-01-06 07:38:15 - Error: Build failed!`

wukongdaily, 2025-01-06:
嗯看到了。
首先建议你fork我的项目。这样我修改后 你可随时sync fork代码
上述问题现在应该不存在了。因为项目里采用的是 `immortalwrt/imagebuilder:x86-64-openwrt-24.10`
而`x86-64-openwrt-24.10` 这个标签 我发现它是变化的，它的特点是，有时候会使用24.10-SNAPSHOT 版本来编译。有时候会使用24.10.0-rc3 来编译。当使用24.10-SNAPSHOT 版本来编译的时候，就有一定概率编译失败，而现在又好了。我什么都没动。我这边打算改回去。还是用rc3版本这种正式版。固定下来。
具体来说就是。我会将
`immortalwrt/imagebuilder:x86-64-openwrt-24.10`
改成
`immortalwrt/imagebuilder:x86-64-openwrt-24.10.0-rc3`

你那边如果不是fork的，你也可以手动修改一下yml中的docker标签。

wukongdaily, 2025-01-06:
[image]

wukongdaily, 2025-01-06:
https://github.com/wukongdaily/AutoBuildImmortalWrt/commit/849504b99701a1bffb051cce974254ff36ac8ca5

closed 2025-01-06 by wukongdaily