wukongdaily/ImmortalWrt-ImageBuilder #25: 建议 修改 building immortalWrt 时

Wang-Mer, 2025-01-08:
```
      - name: Get current date and time
        id: date
        run: echo "current_date=$(date '+%Y年%m月%d日-%H时%M分')" >> $GITHUB_ENV

      - name: Create GitHub Release Info
        uses: softprops/action-gh-release@v2.1.0
        with:
          tag_name: Autobuild-x86-64
          name: "ImmortWrt-[那个版本(24.10 或 23.05.4)]-x86-64-efi ${{ env.current_date }}"
          body_path: ${{ github.workspace }}/info.md
          draft: false
          prerelease: false
        env:
          GITHUB_TOKEN: ${{ secrets.GITHUB_TOKEN }}
```

Wang-Mer, 2025-01-08:
如果 编译了 24.10 然后去编译 23.05.4 就分不清那个是那个了

wukongdaily, 2025-01-08:
24.10 和 23.05.4 这两条工作流出来的文件名是不同的呀。能做出区分的。

`immortalwrt-24.10.0-rc3-x86-64-generic-squashfs-combined-efi.img.gz`

`immortalwrt-23.05.4-x86-64-generic-squashfs-combined-efi.img.gz`

Wang-Mer, 2025-01-08:
ok，没注意他们已经和在一起了

closed 2025-01-08 by Wang-Mer