wukongdaily/ImmortalWrt-ImageBuilder #42: 我想要Sirpdboy这个“高级设置”插件-如何加入到固件？

androidcn, 2025-01-13:
我想要Sirpdboy这个“高级设置”插件-如何加入到固件？

https://github.com/sirpdboy/luci-app-advancedplus

wukongdaily, 2025-01-15:
你直接安装ipk就行。检查一下 如果有luci-app-argon-config 先移除
`opkg remove luci-app-argon-config --force-removal-of-dependent-packages`

然后你在安装就行了。比如x86-64的
```
wget -O luci-app-advancedplus.ipk https://dl.openwrt.ai/packages-24.10/x86_64/kiddin9/luci-app-advancedplus_25.335.56274~b1cf551_all.ipk
opkg install luci-app-advancedplus.ipk
```

androidcn, 2025-01-15:
> 你直接安装ipk就行。检查一下 如果有luci-app-argon-config 先移除
>
> `opkg remove luci-app-argon-config --force-removal-of-dependent-packages`
>
>
>
> 然后你在安装就行了。比如x86-64的
>
> ```
>
> wget -O luci-app-advancedplus.ipk [...]

感谢帮我找到ipk…immortal的内置的软件源没有这个ipk

wukongdaily, 2025-01-16:
哈哈 我们要感谢万能的K大。他编译了许多ipk。而且像这类没什么依赖的ipk 可以直接安装。

closed 2025-01-16 by wukongdaily