wukongdaily/ImmortalWrt-ImageBuilder #43: 24.10版的固件在R4S跑不起来

choicky, 2025-01-13:
多谢大佬的工作流脚本，让我也能编译固件了😁

我发现23.05的固件在R4S跑得很好，但24.10在R4S跑不起来。不知道原因在哪。

有没有网友遇到类似的问题？如何解决的？

wukongdaily, 2025-01-15:
是吗～ 有没有具体的固件的地址呢。我这边也试试

choicky, 2025-01-15:
我编译的固件在这里：  https://github.com/choicky/AutoBuildImmortalWrt/releases

24.10的，我编译的几个都没办法跑R4S 

此外，我搜到 https://github.com/immortalwrt/immortalwrt/issues/1593 有人说最近半年immortalwrt的24.10固件都没办法启动R4S

wukongdaily, 2025-01-16:
确实跑不起来，rc1 连dhcp都没有，rc2 能分配ip 网页能出现一会。rc3 也是能dhcp，出不来网页，
总之这三个都不行。我估计跟内核版本有关系。你看下这位大佬编译的吧。。它的内核是6.12.x 这版本就没问题。rc3我试过，rc4、5 应该也可以。
https://github.com/sbwml/builder/releases

wukongdaily, 2025-01-16:
然后你可以用这个项目来编译 https://github.com/wkccd/r4s_build_script/blob/master/openwrt/24-config-common 自己删减插件什么的。

closed 2025-01-17 by wukongdaily

putaomeiyoupi, 2025-02-09:
我的R4S刷了immortalwrt的24.10，也是起不起来。现在刷到23.05.4的，都不提示升级24.10了。
我是昨天刷的，折腾了半天又刷回去了。

Yeeeeeer, 2025-02-09:
对的，我不管用24的RC还是24.10，immortalwrt原版，还是大佬的编译版本，都起不起来，开机几分钟全部等常亮类似死机的状态。

choicky, 2025-02-10:
> 确实跑不起来，rc1 连dhcp都没有，rc2 能分配ip 网页能出现一会。rc3 也是能dhcp，出不来网页， 总之这三个都不行。我估计跟内核版本有关系。你看下这位大佬编译的吧。。它的内核是6.12.x 这版本就没问题。rc3我试过，rc4、5 应该也可以。 https://github.com/sbwml/builder/releases

补充反馈一下。
我后来尝试过大佬推荐的这款。能跑起来，但是，web访问路由器的管理后台时，感觉卡卡的、反应迟钝，不喜欢。
我的R4S暂时跑 23.03了。

不是说R4S进入了openwrt官方支持列表了么？怎么24.10反而挂了那么久

mickeyman115, 2025-02-10:
同样的问题。比较新版本的固件都跑不起来，包括immortal 23. 05，需要刷写21版本的老固件才能亮sys，然后再刷23.05才能正常工作。刷24.10直接死机变砖。这破烂玩意太闹心。

choicky, 2025-02-10:
> 同样的问题。比较新版本的固件都跑不起来，包括immortal 23. 05，需要刷写21版本的老固件才能亮sys，然后再刷23.05才能正常工作。刷24.10直接死机变砖。这破烂玩意太闹心。

我之前编译的 23.05 能跑R4S。如感兴趣可去下载试试
https://github.com/choicky/AutoBuildImmortalWrt/releases