wukongdaily/ImmortalWrt-ImageBuilder #48: 麻烦请教下，如何添加直接编译esxi的版本，这样比较方便？

hbzmt, 2025-01-16:
麻烦请教下，如何添加直接编译esxi的版本，这样比较方便？

wukongdaily, 2025-01-16:
这需要你稍微改一下配置
项目里 x86-64/imm.config 此文件中
`CONFIG_VMDK_IMAGES=n `改成y

而且yml尾部输出文件那边也需要增加一行代码

`${{ github.workspace }}/bin/targets/x86/64/*.vmdk`

```
- name: Upload ImmortWrt as release assets
        uses: softprops/action-gh-release@v2.2.1
        with:
          tag_name: Autobuild-x86-64
          body_path: ${{ github.workspace }}/info.md
          files: |
            ${{ github.workspace }}/bin/targets/x86/64/*.vmdk
            ${{ github.workspace }}/immortalwrt-23.05.4-x86-64-generic-squashfs-combined-efi.img.gz
            ${{ github.workspace }}/immortalwrt-23.05.4-x86-64-generic-squashfs-combined-efi.img.gz.sha256
          token: ${{ secrets.GITHUB_TOKEN }}
        env:
          GITHUB_TOKEN: ${{ secrets.GITHUB_TOKEN }}
```

hbzmt, 2025-01-16:
谢谢大佬的回复，如何加入adguardhome插件，我试过在软件包处添加，未成功。
另是否可以加入固件更新功能？

wukongdaily, 2025-01-16:
找个我回头做成run包。

hbzmt, 2025-01-16:
> 这个我回头做成run包。
我没装istore run包没法安装吧

closed 2025-01-17 by wukongdaily