wukongdaily/ImmortalWrt-ImageBuilder #49: 这个编译的固件能不能使用-值守式系统更新，这个貌似是immortalWrt的特色，可以专题讲讲

androidcn, 2025-01-16:
这个编译的固件能不能使用-值守式系统更新，这个貌似是immortalWrt的特色，可以专题讲讲

luci-app-attendedsysupgrade

wukongdaily, 2025-01-20:
提到升级，就不得不提到服务器托管。如果 值守式系统 存了一个300M的固件，而假设我们的固件是2G。2G的固件若升级，会怎样。回头我先测测吧。如果它能做到 “差分升级” 那还值得一说。如果 从大 升级 到小 那就得不偿失了。

androidcn, 2025-01-27:
> 提到升级，就不得不提到服务器托管。如果 值守式系统 存了一个300M的固件，而假设我们的固件是2G。2G的固件若升级，会怎样。回头我先测测吧。如果它能做到 “差分升级” 那还值得一说。如果 从大 升级 到小 那就得不偿失了。

rc4出来了…值守式出错

服务器响应：Error: Impossible package selection

```
* opkg_install_cmd: Cannot install package app-meta-ddnsgo.
 * opkg_install_cmd: Cannot install package app-meta-nlbwmon.
 * opkg_install_cmd: Cannot install package app-meta-sftp.
 * opkg_install_cmd: Cannot install package luci-app-advancedplus.
 * opkg_install_cmd: Cannot install package luci-app-mihomo.
 * opkg_install_cmd: Cannot install package luci-app-store.
 * opkg_install_cmd: Cannot install package luci-i18n-mihomo-zh-cn.
 * opkg_install_cmd: Cannot install package luci-i18n-quickstart-zh-cn.
 * opkg_install_cmd: Cannot install package luci-lib-taskd.
 * opkg_install_cmd: Cannot install package mihomo.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2
```

wukongdaily, 2025-02-11:
```
Pseudo file "dev" exists in source filesystem "/builder/build_dir/target-x86_64_musl/root-x86/dev".
Ignoring, exclude it (-e/-ef) to override.
error: ext4_allocate_best_fit_partial: failed to allocate 4171 blocks, out of space?
make[3]: *** [/builder/include/image.mk:399: /builder/build_dir/target-x86_64_musl/linux-x86_64/root.ext4] Error 1
```
这个值守式系统 不适合咱们。我看了 它其实还是利用云端的imagebuilder来构建的，并且还是有空间限制的。
三个字形容就是不实用。尤其你如果安装的是1GB的固件，它升级的时候也要先构建本地已有的配置，但是它因为地方不够，它就构建不了。
总而言之，它只适合 能接受默认300M固件大小的用户。不适合咱们这种“大固件”

最后回答你的问题
直接「保留配置」 升级固件可行吗？

我在视频里多次强调 所谓的升级固件 其实就相当于换了一个固件，你要确保新固件的大小和原固件大小一致。或者新固件比原固件要大，不能比它小。如果新的固件大小 >= 原固件，那么你保留配置 也许是可行的。但是如果从大到小 肯定不能保留配置的。

closed 2025-02-11 by wukongdaily