wukongdaily/ImmortalWrt-ImageBuilder #53: 关于创建docker immortalwrt的疑问

cjlhll, 2025-01-17:
您好，悟空，我用N1根据教程在armbian的docker里部署了immortalwrt，但是我有个疑问，就是下面这段代码，的66.2 ip是干啥的，这个ip需要在局域网内没有别的设备使用是吗？我设置了一个没有设备使用的ip后，我查询不到这个ip，追溯不到了，是干什么的？我看追溯不到，我把immortalwrt的静态ip设置成这个就打不开了。
`
ip link add macvlan-shim link eth0 type macvlan mode bridge
ip addr add 192.168.66.2/24 dev macvlan-shim
ip link set macvlan-shim up
`

我现在是1是主路由，2是armbian，我把这个ip设置为3了，然后immortalwrt设置为了4，现在就搞不懂这个3在哪里。

cjlhll, 2025-01-17:
现在ping 3也ping不通。

wukongdaily, 2025-01-19:
宿主机已经占用了 192.168.5.3 这个 IP 地址。如果你在 Docker 容器中使用相同的网络段（192.168.5.0/24），你应该避免分配给容器 192.168.5.3 这个地址，因为它已经被宿主机使用了。

第二句 ip addr add 192.168.5.3/24 dev macvlan-shim 的作用是将 IP 地址 192.168.5.3 和子网掩码 /24 配置到名为 macvlan-shim 的虚拟网络接口上。

closed 2025-01-20 by wukongdaily