wukongdaily/ImmortalWrt-ImageBuilder #66: 添加插件后编译出现“系统上存在旧版 iptables 规则。 不鼓励混合使用 iptables 和 nftables 规则，这可能会导致流量过滤不完整。”

magi-Z, 2025-01-25:
试着编译个大全版加入这些插件，是这些插件里的防火墙用的不一样吗？

PACKAGES="$PACKAGES luci-i18n-alist-zh-cn"
PACKAGES="$PACKAGES luci-i18n-netdata-zh-cn"
PACKAGES="$PACKAGES luci-i18n-qos-zh-cn"
PACKAGES="$PACKAGES luci-i18n-arpbind-zh-cn"
PACKAGES="$PACKAGES luci-i18n-mwan3-zh-cn"
PACKAGES="$PACKAGES luci-i18n-mwan3helper-zh-cn"
PACKAGES="$PACKAGES luci-i18n-ramfree-zh-cn"
PACKAGES="$PACKAGES luci-i18n-rclone-zh-cn"
PACKAGES="$PACKAGES luci-i18n-samba4-zh-cn"
PACKAGES="$PACKAGES luci-i18n-smartdns-zh-cn"
PACKAGES="$PACKAGES luci-i18n-statistics-zh-cn"
PACKAGES="$PACKAGES luci-i18n-ddns-go-zh-cn"
PACKAGES="$PACKAGES luci-i18n-usb-printer-zh-cn"
PACKAGES="$PACKAGES luci-i18n-watchcat-zh-cn"
PACKAGES="$PACKAGES luci-i18n-timewol-zh-cn"
PACKAGES="$PACKAGES luci-i18n-appfilter-zh-cn"
PACKAGES="$PACKAGES luci-i18n-arpbind-zh-cn"
PACKAGES="$PACKAGES luci-i18n-cifs-mount-zh-cn"
PACKAGES="$PACKAGES luci-i18n-eqos-zh-cn"
PACKAGES="$PACKAGES luci-i18n-hd-idle-zh-cn"
PACKAGES="$PACKAGES luci-i18n-minidlna-zh-cn"
PACKAGES="$PACKAGES luci-i18n-smartdns-zh-cn"
PACKAGES="$PACKAGES luci-i18n-wol-zh-cn"
PACKAGES="$PACKAGES luci-i18n-zerotier-zh-cn"
PACKAGES="$PACKAGES luci-i18n-vlmcsd-zh-cn"
PACKAGES="$PACKAGES luci-i18n-upnp-zh-cn"
PACKAGES="$PACKAGES luci-i18n-mwol-zh-cn"
PACKAGES="$PACKAGES luci-i18n-adbyby-plus-zh-cn"
PACKAGES="$PACKAGES luci-i18n-ahcp-zh-cn"
PACKAGES="$PACKAGES luci-i18n-ddns-zh-cn"
PACKAGES="$PACKAGES luci-i18n-cpulimit-zh-cn"
PACKAGES="$PACKAGES luci-i18n-cifs-mount-zh-cn"
PACKAGES="$PACKAGES luci-i18n-base-zh-cn"
PACKAGES="$PACKAGES luci-i18n-aria2-zh-cn"
PACKAGES="$PACKAGES luci-i18n-adblock-fast-zh-cn"
PACKAGES="$PACKAGES luci-i18n-oled-zh-cn"
PACKAGES="$PACKAGES luci-i18n-uhttpd-zh-cn"
PACKAGES="$PACKAGES luci-i18n-gost-zh-cn"
PACKAGES="$PACKAGES luci-i18n-udpxy-zh-cn"
PACKAGES="$PACKAGES luci-i18n-wechatpush-zh-cn"
PACKAGES="$PACKAGES luci-i18n-wifischedule-zh-cn"
PACKAGES="$PACKAGES luci-i18n-nlbwmon-zh-cn"
PACKAGES="$PACKAGES luci-i18n-socat-zh-cn"

wukongdaily, 2025-01-25:
这有点复杂了。不确定具体哪个插件用的是旧版防火墙。

gregorygood, 2025-01-26:
mwan3

wukongdaily, 2025-02-05:
弄少点一个个排除吧。

closed 2025-02-05 by wukongdaily

free-diy, 2025-02-07:
docker问题，把docker去掉就没这问题了

sde31c, 2025-02-16:
docker的问题，docker还在用iptables，打包固件的时候如果选择了docker，就把iptables-nft和ip6tables-nft同时打包进去就可以了。