wukongdaily/ImmortalWrt-ImageBuilder #94: 24.10版已经正式发布，但编译的固件仍然是RC4

griphin, 2025-02-10:
如题

wukongdaily, 2025-02-11:
不会啊。我这边编译出来就是24.10.0 呀。如图
你看看你的代码是最新的吗。
在构建过程里，你看看那个imagebuilder 写的标签是不是rc4？

[image]
[image]

griphin, 2025-02-11:
4天前编译的，编译之前更新了代码，但是编译出来的固件，文件名有RC4标签。我咋没找到您截图上的build界面在哪里。

[image]

wukongdaily, 2025-02-11:
要不你把项目删除了，从新 Fork 一次试试。因为我这边都试过了 是正常的。你可以在我的release里看到的。不带rc4的字样。

griphin, 2025-02-11:
感谢大佬指点，我来删掉这个项目，重新fork下

griphin, 2025-02-11:
> 要不你把项目删除了，从新 Fork 一次试试。因为我这边都试过了 是正常的。你可以在我的release里看到的。不带rc4的字样。

按照大佬的指示，删掉项目，重新fork，果然编译结果里没有RC4了。 请问有没有办法把immortalwrt软件库里之外的插件给集成进去？Lucky是个十分优秀的插件，可以提供webdav挂载，动态域名，自动申请证书等等需要多个插件才能提供的功能，可惜每次刷完固件，都要手动上传三个IPK文件，非常不方便，要是能直接集成进去就好了。

wukongdaily, 2025-02-12:
`把immortalwrt软件库里之外的插件给集成进去`
这目标对于目前我的水平来说 还是有点难度的。理论上imagebuilder 是能集成外部ipk的，
但是比较费劲，因为它需要生成一个index索引，就是一种依赖关系列表。让imagebuilder来识别。
我觉得做肯定是能做的 只是前无古人 比较麻烦而已。我抽空会研究一下 这个“主题”

目前我能想到的一个办法就是用另一个项目来生成run文件 https://github.com/wukongdaily/RunFilesBuilder
用户只需要一个run文件，执行它 就能把相关的ipk 全部安装完毕。
目前我正在完善此项目，做了7个插件的合成了。我当然也能合成一个Lucky 这工作比上面的主题要容易。
你可以看看此项目。

griphin, 2025-02-12:
感谢回复，不过我现在还是linux小白，还没搞明白这个run到底是干嘛的，所以RunFilesBuilder这个项目明显有点超纲。另外，从build.sh里看到了iStore的字样，难道immortalwrt里能安装iStore?

wukongdaily, 2025-02-13:
是的 仔细看看吧。run就是一个自解压程序。把ipk都打包在一起了。这样用户就不需要在考虑 要安装哪些依赖了。
immortalwrt里能安装iStore？是的。任何x86-64 和 arm a53平台的openwrt都能安装istore

griphin, 2025-02-13:
老大出一期怎么在immortalwrt安装istore插件的教程吧，要知道immortalwrt是个极好的固件，就是作者太犟，官网默认的软件空间是300M就不说了，关键是插件仓库实在太过保守，很多小伙伴们喜闻乐见的插件，譬如lucky和qibittorrent-simple (自带的qb不能指定配置文件位置），就是不给收录，隔壁Kiddin9大神的Kwrt软件仓库极其丰富，定制选项也非常多，就是固件稳定性很差。我曾经把immortwrt的软件仓库改成kwrt的，但是出现了一些问题。哦，跑题了，主要istore找不到现成的插件，就算找到了x86架构的luci-app-istore.ipk文件，也不知道是否还需要什么依赖。要是istore能集成进去，就能大大缓解immortalwrt软件太过保守的弊端。

wukongdaily, 2025-02-14:
最最最主要的问题是什么？是你不能每期视频都看我的视频。真是伤心太平洋了我。《immortalwrt安装istore插件的教程》这都讲过啊。再说文章都给你了。

【香橙派Zero3上体验ImmortalWrt ｜附基本用法｜旁路｜轻NAS｜docker｜iStore安装等】 【精准空降到 04:04】 https://www.bilibili.com/video/BV1MarQYMEJD/?share_source=copy_web&vd_source=0bb92241fb28a55c32c2e5132116b594&t=244

closed 2025-02-14 by wukongdaily

griphin, 2025-02-14:
> 最最最主要的问题是什么？是你不能每期视频都看我的视频。真是伤心太平洋了我。《immortalwrt安装istore插件的教程》这都讲过啊。再说文章都给你了。
> 
> 【香橙派Zero3上体验ImmortalWrt ｜附基本用法｜旁路｜轻NAS｜docker｜iStore安装等】 【精准空降到 04:04】 https://www.bilibili.com/video/BV1MarQYMEJD/?share_source=copy_web&vd_source=0bb92241fb28a55c32c2e5132116b594&t=244

额.....我确实在油管上看到过这个视频标题，但是一看到“香橙派”我就直接忽略了后面的内容，我又没有这个设备。[手动尴尬]