wukongdaily/ImmortalWrt-ImageBuilder #107: 使用此工作流编译X86/64平台，24.10固件，目前出现的2个已知问题

griphin, 2025-02-23:
1. 不做改变任何默认设置，一路next下去，可以正确编译，修改build.sh里的插件后编译，固件大小2048，DOCKER：是， PPPOE: 是，进度到dnsmasq-full那里的时候，就会报错，即便把所有添加的插件全部注释掉，仍然会报错。
Configuring dnsmasq-full.
make: *** [Makefile:337: image] Error 2
2025-02-23 15:11:15 - Error: Build failed!
Error: Process completed with exit code 1

2. 进入luci后台，点击状态->防火墙，会有提示：系统上存在旧版 iptables 规则。 不鼓励混合使用 iptables 和 nftables 规则，这可能会导致流量过滤不完整。 至于有什么影响，暂时还没有发现。

.

griphin, 2025-07-26:
> 第2个问题，有办法处理吗？
> 
> <img alt="Image" width="1009" height="130" src="https://private-user-images.githubusercontent.com/22655724/470193192-24b5c13b-2ba0-49d7-bdc5-a1a27ecfe2e8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTM0OTU3NDYsIm5iZiI6MTc1MzQ5NTQ0NiwicGF0aCI6Ii8yMjY1NTcyNC80NzAxOTMxOTItMjRiNWMxM2ItMmJhMC00OWQ3LWJkYzUtYTFhMjdlY2ZlMmU4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzI2VDAyMDQwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYxM2MzNDk4NzViMWFiZjI3YzU4OTA0MGM4NTE2ZGQ2ZjA5ZjA1Yzc4YTc5MjJjOGM1M2JmNWNlYzQzMDZiZTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.mDdp-bWxySqPjt1ategUgy6_C5tTlz_24oITIk3TpZM">

我已经找到了根源，不要编译docker，就不会有这个问题，我猜想docker仍然使用的是iptables

xiealon, 2026-09-13:
iptables那个不用管 因为docker默认的是iptables