wukongdaily/ImmortalWrt-ImageBuilder #112: 没有docker端口

eaglehhha, 2025-02-28:
R2s安装固件，docker没有网络端口，docker不能正常工作

wukongdaily, 2025-03-03:
截个图看看？docker info 有正常返回没

eaglehhha, 2025-03-04:
> 截个图看看？docker info 有正常返回没

忘记截图了，重新去它的官网下载了，按照你以前的教程，以前的教程也是可以添加各种插件，但只要使用了首次运行时启用的脚本，就会出现各种问题（在我的r2s上）。

[image]就是这里没有

eaglehhha, 2025-03-04:
> > 截个图看看？docker info 有正常返回没
> 
> 忘记截图了，重新去它的官网下载了，按照你以前的教程，以前的教程也是可以添加各种插件，但只要使用了首次运行时启用的脚本，就会出现各种问题（在我的r2s上）。
> 
> [image]就是这里没有

如果自己新建接口，会显示不存在的设备接口

eaglehhha, 2025-03-07:
ImmortalWrt 24.10.0, r32824-6a73dae98c9c
 -----------------------------------------------------
root@ImmortalWrt:~# docker info
Client:
 Version:    27.3.1
 Context:    default
 Debug Mode: false
 Plugins:
  compose: Docker Compose (Docker Inc.)
    Version:  v2.30.1
    Path:     /usr/lib/docker/cli-plugins/docker-compose

Server:
 Containers: 0
  Running: 0
  Paused: 0
  Stopped: 0
 Images: 0
 Server Version: 27.3.1
 Storage Driver: vfs
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Cgroup Version: 2
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local splunk syslog
 Swarm: inactive
 Runtimes: io.containerd.runc.v2 runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version:
 runc version:
 init version: de40ad0
 Security Options:
  seccomp
   Profile: builtin
  cgroupns
 Kernel Version: 6.6.73
 Operating System: ImmortalWrt 24.10.0
 OSType: linux
 Architecture: aarch64
 CPUs: 4
 Total Memory: 978.8MiB
 Name: ImmortalWrt
 ID: 37ca123e-800d-45f9-b53a-942c97a715c7
 Docker Root Dir: /opt/docker
 Debug Mode: false
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false

WARNING: No swap limit support

liuweilhy, 2025-03-08:
我测试在x86-64 24.10下，docker的启动脚本似乎不会在新固件首次运行时正确执行，看不到docker接口。此时重启路由器使docker脚本正确执行就能看到接口了。也就是说，刷入新固件，开机启动之后要再重启一次。
另外，如果容器不能访问，可以试试防火墙里将docker区域的转发规则中的“转发到”和“来自于”把lan和wan都加入进去。

eaglehhha, 2025-03-08:
> 我测试在x86-64 24.10下，docker的启动脚本似乎不会在新固件首次运行时正确执行，看不到docker接口。此时重启路由器使docker脚本正确执行就能看到接口了。也就是说，刷入新固件，开机启动之后要再重启一次。 另外，如果容器不能访问，可以试试防火墙里将docker区域的转发规则中的“转发到”和“来自于”把lan和wan都加入进去。

感谢，重启后docker接口出来了

jianyun19999, 2025-03-17:
> > 我测试在x86-64 24.10下，docker的启动脚本似乎不会在新固件首次运行时正确执行，看不到docker接口。此时重启路由器使docker脚本正确执行就能看到接口了。也就是说，刷入新固件，开机启动之后要再重启一次。 另外，如果容器不能访问，可以试试防火墙里将docker区域的转发规则中的“转发到”和“来自于”把lan和wan都加入进去。
> 
> 感谢，重启后docker接口出来了

重启后docker的接口出来也没有用，通不了，只有这个办法是正解，需要手动在防火墙里把docker的lan和wan加到转发里

wukongdaily, 2025-05-31:
OK 问题修复。在虚拟机和物理机测试通过✅ 
https://github.com/wukongdaily/AutoBuildImmortalWrt/commit/02aa07b745bc3aabc58ef150eeed226f933fa08b