wukongdaily/ImmortalWrt-ImageBuilder #118: FileBrowse可用

KEVINJ1E, 2025-03-07:
x86_64 24.10.0版本编译完成，FileBrowser开启，web显示404 Not Found，呈不可用状态。

wukongdaily, 2025-03-07:
https://github.com/wukongdaily/AutoBuildImmortalWrt/commit/56f3cd8ed5a8643034278eb765610e4aa43c64bc
24.10 下应该用这个文件管理器。它在【服务】菜单下。用户名 和密码是admin

wukongdaily, 2025-03-07:
luci-i18n-filebrowser-go-zh-cn

wukongdaily, 2025-03-07:
这不是有嘛～ 就是FileBrowser （luci-i18n-filebrowser-go-zh-cn）就是它 。不是文件管理器。文件管理器这款app 不适合24.10 因为UI会错乱的。所以我才换成这款。然后23.05 可以安装文件管理器的。明白了吧

KEVINJ1E, 2025-03-07:
[image]

[image]

fengtangyuan, 2025-03-10:
用值守式更新编译新固件后可以打开，用流编译的新固件就不行，感觉很迷

fengtangyuan, 2025-03-11:
FileBrowser软件包管理中版本已经退回到2.31.2，卸载重新安装后即可解决

KEVINJ1E, 2025-03-11:
> FileBrowser软件包管理中版本已经退回到2.31.2，卸载重新安装后即可解决

已测试，可用。

wukongdaily, 2025-05-09:
OK 在24.10.1 测试通过 目前可用了