wukongdaily/ImmortalWrt-ImageBuilder #142: 总结遇到的问题

jianyun19999, 2025-03-18:
1. 防火墙报错，显示两个防火墙无法叠加的报告
2. 使用Nikki时在导入定阅后更新面板，然后openwrt整个卡死，无法用网页连接，可以用终端连接进行REBOOT
3. 使用Docker时，没有防火墙配置，局域网无法登陆
4. 使用Docker时无法穿透端口映射，DNS服务器改成0了依然无效

jianyun19999, 2025-03-19:
虽然有这样哪样的问题，但还是这个固件好用

jianyun19999, 2025-03-19:
在软件包删除软件时出现这个错误 
XHRrequestabortedbybrowser

AXY520, 2025-06-13:
问下大哥docker局域网无法访问这个问题，防火墙具体怎么配置的

wukongdaily, 2025-06-14:
> 问下大哥docker局域网无法访问这个问题，防火墙具体怎么配置的

此问题 已经解决了。更新代码后 从新构建就好了。https://github.com/wukongdaily/AutoBuildImmortalWrt/commit/02aa07b745bc3aabc58ef150eeed226f933fa08b

AXY520, 2025-06-14:
> > 问下大哥docker局域网无法访问这个问题，防火墙具体怎么配置的
> 
> 此问题 已经解决了。更新代码后 从新构建就好了。[02aa07b](https://github.com/wukongdaily/AutoBuildImmortalWrt/commit/02aa07b745bc3aabc58ef150eeed226f933fa08b)

昨天新构建的，好像没行，docker里部署了一个订阅转换后端，没办法用，但是我看防火墙里确实有docker转发

closed 2025-07-13 by wukongdaily