wukongdaily/ImmortalWrt-ImageBuilder #154: 23.05.4编译r2s出错

grayshall, 2025-04-02:
Configuring dockerd.
Configuring uci.
Configuring odhcpd-ipv6only.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency shadowsocks-rust-sslocal for luci-app-passwall
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-passwall found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-passwall-zh-cn:
 * 	shadowsocks-rust-sslocal
 * opkg_install_cmd: Cannot install package luci-i18n-passwall-zh-cn.
Configuring ipv6helper.
Configuring dropbear.

使用的未修改imm.config，一直报错如上。请问如何解决，谢谢！

grayshall, 2025-04-02:
24.10.0编译r2s正常

wukongdaily, 2025-04-03:
嗯 我也发现了 不过应该不是我这里的错误。因为网页上 也同样报错。估计就需要天灵他们去修改了。

[image]

wukongdaily, 2025-05-04:
现在应该正常了

closed 2025-05-04 by wukongdaily