wukongdaily/ImmortalWrt-ImageBuilder #168: 如何添加防火墙自定义规则

chfjxmtable, 2025-04-27:
找不到防火墙自定义规则添加的地方，不知从何处下手。请指教，谢谢！

tomy2502, 2025-05-02:
AutoBuildImmortalWrt/files/etc/uci-defaults/99-custom.sh

这里面加，我加了些 端口映射

uci add firewall redirect
uci set firewall.@redirect[-1].name='myfrp-1'
uci set firewall.@redirect[-1].src='wan'
uci set firewall.@redirect[-1].src_dport='7000'
uci set firewall.@redirect[-1].dest='lan'
uci set firewall.@redirect[-1].dest_ip='192.168.100.1'
uci set firewall.@redirect[-1].dest_port='7000'
uci set firewall.@redirect[-1].proto='tcp'
uci set firewall.@redirect[-1].target='DNAT'

chfjxmtable, 2025-05-06:
> uci set ***@***.***[-1].name='myfrp-1'
> uci set ***@***.***[-1].src='wan'
> uci set ***@***.***[-1].src_dport='7000'
> uci set ***@***.***[-1].dest='lan'
> uci set ***@***.***[-1].dest_ip='192.168.100.1'
> uci set ***@***.***[-1].dest_port='7000'
> uci set ***@***.***[-1].proto='tcp'
> uci set ***@***.***[-1].target='DNAT'

谢谢指教！

closed 2025-05-31 by wukongdaily