wukongdaily/ImmortalWrt-ImageBuilder #211: 构建chiprock r4se固件不成功

magi-Z, 2025-07-13:
每次最后出现如下：
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-wechatpush-zh-cn.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:161: _call_image] Error 2
make: *** [Makefile:337: image] Error 2
2025-07-12 15:34:46 - Error: Build failed!
Error: Process completed with exit code 1.

wukongdaily, 2025-07-13:
# 包名写错了
## 你的错误日志截取错啦 ，下次往上面找 就能找到真实的报错信息 末尾往往不是。
## 近期alist 变更为openlist 正确的包名应该是 `luci-i18n-openlist-zh-cn`
```
Collected errors:
 * opkg_install_cmd: Cannot install package luci-i18n-alist-zh-cn.
```
## 如何查询imm仓库内有哪些插件
https://mirrors.sjtug.sjtu.edu.cn/immortalwrt/releases/24.10.2/packages/x86_64/luci/

magi-Z, 2025-07-13:
编译前同步 fork项目也不行呀？

wukongdaily, 2025-07-13:
不是啊  luci-i18n-alist-zh-cn 这应该是你自己添加的包，不是项目默认的。我看到你已经构建成功了。

closed 2025-07-13 by wukongdaily