wukongdaily/ImmortalWrt-ImageBuilder #238: ext4格式固件生成错误

hbzmt, 2025-07-22:
按照之前和你的电邮沟通，将yaml文件里的文件格式替换成了*ext4-combined-efi.img.gz，之前编译一直正常，今天按照你最新的视频，更新了之后，也把yaml文件修改了，编译不成功，如下错误，麻烦帮忙看下。我需要EXT4格式的固件。谢谢！

Run cp /home/runner/work/AutoBuildImmortalWrt/AutoBuildImmortalWrt/bin/targets/x86/64/*ext4-combined-efi.img.gz /home/runner/work/AutoBuildImmortalWrt/AutoBuildImmortalWrt
cp: cannot stat '/home/runner/work/AutoBuildImmortalWrt/AutoBuildImmortalWrt/bin/targets/x86/64/*ext4-combined-efi.img.gz': No such file or directory
Error: Process completed with exit code 1.

wukongdaily, 2025-07-23:
我看你的工作流已经成功的生成的ext4 格式的固件呀，这问题是不是目前不存在了？
https://github.com/hbzmt/AutoBuildImmortalWrt/releases/tag/Autobuild-x86-64

hbzmt, 2025-07-23:
> 我看你的工作流已经成功的生成的ext4 格式的固件呀，这问题是不是目前不存在了？ https://github.com/hbzmt/AutoBuildImmortalWrt/releases/tag/Autobuild-x86-64

imm.config里的ext4要改成y，昨晚弄好了。另外请教个问题，我只选了passwall和NIKKI，用final ssh登录进去，无法像别的固件一样可以定位到文件，这样就无法上传NIKKI的某些文件，这个是不是要选则添加一些插件才行。我用其他的版本编译的Immortalwrt是可以的，谢谢

closed 2025-07-24 by wukongdaily

hbzmt, 2025-08-01:
> 要安装 `openssh-sftp-server` 这个插件才行。 istore商店下载 或者 软件包 搜索下载都行。或者构建的时候 带上这个包。shell/custom-packages.sh 这个里面 不仅仅可以放仓库外的包，也能放仓库内的包，它说白了仅仅就是字符串的拼接。

谢谢大佬的解答