wukongdaily/ImmortalWrt-ImageBuilder #268: x86 24.10编译的，系统软件包没翻译过来，还是显示英文，其他目前都正常。

hbzmt, 2025-08-01:
仓库内容选择如下：

# ============= imm仓库内的插件==============
# 定义所需安装的包列表 下列插件你都可以自行删减
PACKAGES=""
PACKAGES="$PACKAGES curl"
#PACKAGES="$PACKAGES luci-i18n-diskman-zh-cn"
PACKAGES="$PACKAGES luci-i18n-firewall-zh-cn"
#PACKAGES="$PACKAGES luci-theme-argon"
#PACKAGES="$PACKAGES luci-app-argon-config"
#PACKAGES="$PACKAGES luci-i18n-argon-config-zh-cn"
#24.10
#PACKAGES="$PACKAGES luci-i18n-package-manager-zh-cn"
#PACKAGES="$PACKAGES luci-i18n-ttyd-zh-cn"
#PACKAGES="$PACKAGES luci-i18n-passwall-zh-cn"
PACKAGES="$PACKAGES luci-app-openclash"
#PACKAGES="$PACKAGES luci-i18n-homeproxy-zh-cn"
PACKAGES="$PACKAGES openssh-sftp-server"
#PACKAGES="$PACKAGES luci-i18n-samba4-zh-cn"
# 文件管理器
#PACKAGES="$PACKAGES luci-i18n-filemanager-zh-cn"
# 静态文件服务器dufs(推荐)
#PACKAGES="$PACKAGES luci-i18n-dufs-zh-cn"
# ======== shell/custom-packages.sh =======
# 合并imm仓库以外的第三方插件
PACKAGES="$PACKAGES $CUSTOM_PACKAGES"

wukongdaily, 2025-08-02:
你没有打开注释 PACKAGES="$PACKAGES luci-i18n-package-manager-zh-cn"  去掉井号就好了

closed 2025-08-02 by wukongdaily

hbzmt, 2025-08-05:
> 你没有打开注释 PACKAGES="$PACKAGES luci-i18n-package-manager-zh-cn" 去掉井号就好了

感谢大佬的回复。