wukongdaily/ImmortalWrt-ImageBuilder #270: imm官网证书到期，导致所有固件构建失败

wukongdaily, 2025-08-03:
https://github.com/immortalwrt/immortalwrt/issues/1995
**等此bug修复。我们就可以用了 大家稍等。**

[image]

[image]

wukongdaily, 2025-08-03:
[image]

kwzzz03, 2025-08-03:
> 出错的都是类似于"https://downloads.immortalwrt.org/releases/24.10.2/targets/x86/64/packages/Packages.gz, "后面是不是多了个“，”，直接用[https://downloads.immortalwrt.org/releases/24.10.2/targets/x86/64/packages/Packages.gz是可以下载的。](https://downloads.immortalwrt.org/releases/24.10.2/targets/x86/64/packages/Packages.gz%E6%98%AF%E5%8F%AF%E4%BB%A5%E4%B8%8B%E8%BD%BD%E7%9A%84%E3%80%82)

这是改哪里才可以编译

wukongdaily, 2025-08-03:
> 出错的都是类似于"https://downloads.immortalwrt.org/releases/24.10.2/targets/x86/64/packages/Packages.gz, [...]

是证书到期了 等官网续上就可以了 咱们不需要做调整的。

wukongdaily, 2025-08-03:
> 这是改哪里才可以编译

哪都不需要动 只需要耐心等待。

wukongdaily, 2025-08-03:
此问题 已解决

closed 2025-08-03 by wukongdaily