wukongdaily/ImmortalWrt-ImageBuilder #376: shell/custom-packages.sh和build.sh有啥区别啊？

cjlhll, 2025-09-01:
悟空您好，我想增减插件的话，应该是改哪个文件啊，我看这俩里都有配置插件呢。

flyhong666, 2025-09-01:
shell/custom-packages.sh  这里面，对照它原来的格式，弄成减号就可以去掉不集成进去。

cjlhll, 2025-09-01:
> shell/custom-packages.sh 这里面，对照它原来的格式，弄成减号就可以去掉不集成进去。

可以输出一个docker用的rootfs文件吗？🤔

cjlhll, 2025-09-01:
> shell/custom-packages.sh 这里面，对照它原来的格式，弄成减号就可以去掉不集成进去。
我看现在N1适配了原生内核呢。能加入进这个项目么

wukongdaily, 2025-09-02:
shell/custom-packages.sh 是项目里 共用的一个脚本。
build.sh 是单个工作流的脚本。

个人使用来说，实际上你修改哪个都行。PACKAGES 这部分 ，你要看懂就会晓得，它只是一个字符串的拼接。最终汇总在一起。

在build.sh 中写道 
```
# 合并imm仓库以外的第三方插件
PACKAGES="$PACKAGES $CUSTOM_PACKAGES"
```

由此可见 你修改哪个都行，最终都是在这里汇总的。

关于N1的问题，因为电视盒子在openwrt中是不被支持的，N1目前是构建armsr-armv8的rootfs 再通过Flippy的打包脚本构建openwrt镜像的。
内核一般是在Flippy打包脚本里设置的。能构建的内核我基本都列出来的。在action的UI列表里选择就行了。内核这东西 是没法在咱们这个项目里控制的。别忘记 这项目的本质是Imagebuilder 它不参与编译。它最多集成第三方应用。